waveform80/pistreaming #1: Connected through ethernet to same switch, still lots of lag / skipped frames

spikedrba, 2015-03-27:
Hi Dave,

thanks a ton for getting this project started, really useful. I'm running into some troubles - happy to do the legwork, but would love some pointers.

Here's what I have:

Pi: B+ running latest raspbian
Desktop: up to date Ubuntu 14.04
Browser: tried latest chrome and FF
Connection: ethernet to the same switch (hp managed, good switch)
Ping flood test:  packets transmitted, 6165 received, 0% packet loss, time 3693ms
rtt min/avg/max/mdev = 0.396/0.555/0.970/0.072 ms, ipg/ewma 0.599/0.510 ms
Other tests: Raspimjpeg and mplayer streaming work just fine with no lag

I'm seeing about 10s lag and some sections of the video are entirely skipped and oveall video plays like in slow motion. Load on the pi is 0.7 while running pistreaming.

Anything I can check or do to further debug the issue?

thanks,

Spike

waveform80, 2015-03-28:
I've played around with this on my Pi 1 and the Pi 2:

On the Pi 1, it seems like lag spikes when there's a significant increase in the amount of info going to/from ffmpeg for transcoding (i.e. if you go from a completely static scene to a high motion scene, ffmpeg suddenly has lots more work to do, there's a spike in the pipes between the application and things lag for a bit until the system settles). If the scene is largely static, or largely high motion for a while, the lag drops back from near the 10s mark to the 1s mark. The Pi 2 seems to handle things much better which isn't too surprising given its got a lot more memory and processing power (the memory is probably more important for this, but I didn't look too closely at the stats while I was running).

It's not at all surprising raspimjpeg is faster - it's doing almost no processing at all on the CPU - just grabbing the raw MJPEG stream straight from the GPU and dumping it over the wire, while pistreaming is grabbing a load of unencoded YUV frames (which are huge), transcoding them with ffmpeg (on the CPU - I'm still amazed the Pi 1 is fast enough for this!), then broadcasting that output to a websockets thread. But then, that's the cost of compatibility; this whole project is basically hack after hack to work around the state of streaming on web clients!

It wouldn't be too difficult to alter this project to use MJPEG output instead (i.e. dump the jsmpeg stuff on the client side and just write a little javascript to pull out individual JPEGs and display them), but then the point of this was to explore what we could do with *video* formats (rather than hacks like MJPEG) - in other words, it's not something I'm terribly interested in doing.

The absolute ideal would be to grab an H.264 stream from the GPU, encapsulate it in an MP4 container, then stream that via websockets; unfortunately I don't know a simple way to do the second bit (encapsulation). I've heard it's possible with a custom ffmpeg build but that exceeds the simplicity I wanted, and it still sounds like a bit of a hack to me (it shouldn't be necessary to shell out to another process - we should be able to do this within the python process). In other words, what I ought to do is go read up on the MP4 format and write the encapsulation myself (it probably isn't *that* hard) - unfortunately, this is way down the list of priorities and not something I've got time to work on. But patches are welcome!

waveform80, 2015-03-28:
Hmm, just noticed something else odd when playing around with this on my Pi 1: pointing the camera at a particularly bland scene (in my case a completely blank, white wall) causes lots of dropped frames. I haven't analyzed this in detail, but my first hunch is that what's happening is that MPEG encoder goes "great!" and compresses the scene *really* well ... too well in fact. So well, that it takes a while for enough frames to fill the 4096 byte chunk that the broadcast thread reads, so the pipeline blocks until a chunk can be filled, but by that time the player has run out of buffered frames and things fall apart when the next chunk comes through. I think this may also explain the lag mentioned above when going from low motion to high motion scenes (the low motion scenes can't fill the broadcast buffer chunk fast enough)

Dropping the buffer size ([here](https://github.com/waveform80/pistreaming/blob/master/server.py#L124)) to 512 bytes seems to fix this quite nicely, but increases the required bandwidth (because there's a lot more broadcasts going on). Might be worth someone experimenting with buffer sizes to find something optimal (512 bytes is just a wild guess on my part)?

The ideal would be a little more complicated (buffer only enough to cover two frames worth of output) but my hunch is that would complicate the code, and this is only meant to be a simple demo for people to build upon.

waveform80, 2015-05-15:
Fixed in 028c1d7ed1

closed 2015-05-15 by waveform80