waveform80/pistreaming #2: You can store the video on disk while streaming?

leonciokof, 2015-03-27:


waveform80, 2015-03-28:
Sure, it wouldn't be difficult to modify the script to write the output of ffmpeg to a disk file in addition to sending it to the websocket. Just modify the loop in `BroadcastThread.run` to write the buffer it receives from ffmpeg out to disk in addition to broadcasting it.

leonciokof, 2015-03-28:
It would be something?

```python
def run(self):
        f = open("video.mpg", "w")
        try:
            while True:
                buf = self.converter.stdout.read(4096)
                if buf:
                    f.write(buf)
                    self.websocket_server.manager.broadcast(buf, binary=True)
                elif self.converter.poll() is not None:
                    break
        finally:
            f.close()
            self.converter.stdout.close()
```

waveform80, 2015-03-28:
That should do the trick (I've made one tiny modification to your patch above, changing the mode from "w" to "wb" - that actually makes no difference on Linux, but it's a good habit to understand when you're writing binary data and when you're writing text).

The one thing to be aware of is that it will be recording to disk *all the time* - even when no clients are connected. When no clients are connected, the `broadcast()` call there does nothing, but the camera is still recording, ffmpeg is still transcoding, and that modification will still write the resulting MPEG data to disk.

closed 2015-09-07 by lgaticaq