waveform80/pistreaming #4: Can we use it for checking latency in video files?

vermagag, 2015-09-07:
Hi, 
Can we use it for checking audio video sync for normal videos ? 
Like to identify there is any latency of audio in specific video of specific video format.

waveform80, 2015-09-07:
Not really - firstly this only streams video, there's no audio. And secondly, even if you did rig it to include audio how would you determine where any de-sync was happening? In the encoder at the server, in the decoder in the client web-browser, in the audio/video sub-systems of whatever OS you were playing on, in the hardware of the client box, the list goes on?

I would've thought that if you were looking for AV desync, the last thing you'd want to do was throw a streaming system (one that transcodes no less) into the middle of things!

vermagag, 2015-09-07:
Thanks for quick response.
I guess my requirement is different from this, I have a bunch of video files, I want to check their audio and video are in synchronization. I thought of writing a script for same. But I'm not getting proper base of that. Do I need to split a video file first into audio and video then I check their respective durations. could you please help me in that.

waveform80, 2015-09-07:
I'd suggest reading up on [ffmpeg](http://ffmpeg.org/) or [libav](http://libav.org/) (they're both from the same original project - I suspect for your purposes there's little to choose between them). They're rather complex systems, but then multimedia is a complex area. Still, you should be able to use either to extract the individual media streams from whatever container format you're dealing with.

vermagag, 2015-09-08:
Okay, Thanks I will surely look forward to it.

closed 2015-09-08 by vermagag