waveform80/pistreaming #5: Frame Rate Issue

doogerk, 2016-02-04:
If the frame rate is lowered below 24, a Broken pipe error happens.

waveform80, 2016-02-05:
Hmm, I can't test right now but for when I get a second, what framerate are you trying? (and I assume you're modifying the framerate by tweaking the FRAMERATE= line at the top of server.py rather than anywhere else?)

doogerk, 2016-02-05:
Yes. I set it to several, but specifically I'm trying somewhere in the 12-15 range. I am using the broadcasting thread to save the data to a file as well and to keep the file size down I want to reduce the  frame rate. Bit rate is more important than frame rate in my case. I did look in the jsmpg.js file and noticed the PICTURE_RATE array does not go below 23.976, so I assume that is the limiting factor.

waveform80, 2016-02-24:
Ah, found it - avconv bombs out with the following error (which you don't see because its output is suppressed):

```
[mpeg1video @ 0x10e7c20] MPEG1/2 does not support 15/1 fps
```

The above was when I attempted 15fps but you get much the same for anything below 24fps. I don't know the MPEG1/2 standards well enough to know what frame rates *are* supported, but I'd hazard a guess that they bottom out at 24fps from what you've described.

doogerk, 2016-02-25:
I've found a usable way around it (at least for my needs). I put an if statement around the broadcast output write to basically do it every 2nd time (increment an integer or alternate a boolean) and also cut the buffer read size in half in the broadcast thread. This effectively records and displays at 12 fps. The video playback speed is 2x though since it still plays at 24 fps. So 1 minute of recording makes a 30 second video. For my purposes this is fine  since the video size is reduced by half and it makes it easier to sift through hours of footage as it runs faster.

closed 2016-02-25 by doogerk

waveform80, 2016-02-25:
Heh - well, it's a cunning method! Glad you found something that'll work for you.