waveform80/pistreaming #6: No Stream on my browser window

4711engel, 2016-02-23:
Hi,
after installing pistreamer on my raspi 2 I started the server     $python server.py   .
After the  information at the console I opened the firefox browser and start the address of my raspi:    http://ip-address:8082/

But I see no stream. Only a grey background.  The two GET lines  show   ... index.html... 200
and    .../jsmpg.js .... 200  .
What goes wrong? Can you help me?   Thanks
uengel@gmx.de

waveform80, 2016-02-23:
Without some sort of error message there's not a great deal I can suggest ... does your camera work normally with raspistill? What's the full output from pistreaming's server script?

4711engel, 2016-02-23:
Hi,
I test the cam with  raspivid -d. No problem.
The message at the LXterminal is:
........ "GET /index.html HTTTP/1.1"  200
........ "GET /jsmpg.js  HTTP/1.1" 200
but the browser window is grey.

4711engel, 2016-02-24:
Testing with other browser I found out that my problem is the firefox browser (V 44). With IE11 streaming works.  But the stream is not disruptively. Every streaming stops after some seconds and continued some seconds later , etc. The gpu_mem = 512 .

waveform80, 2016-02-24:
Hmm, sounds like there may be some issue with your Firefox setup (noscript or something getting in the way?). I've just tried it with Firefox (v44) on my Ubuntu desktop and it's working fine for me. If streaming is intermittent then it's either down to a poor network connection (both my boxes are connected via ethernet) or your Pi is not keeping up with the MPEG1 transcoding (this can happen on Pi 1's - it's very rare on Pi 2's though so I'd suspect the network first in your case, unless you've got other stuff running on your Pi at the same time).

harakiwi, 2016-05-04:
same "problem" here on my end. stream works fine in IE11 and Edge, but Chrome and Firefox (46 and 49) won't display anything (only a grey page). Not really a problem maybe something weird with a media setting. We all know about the fun of standards and videos in websites :/

anyways, thanks for this super easy way of streaming lag-free! :D

harakiwi, 2016-05-04:
sorry for double posting, but the edit button isn't working on epiphany

my log:

pi@raspberrypi:~/pistreaming $ python3 server.py
Initializing camera
Initializing websockets server on port 8084
Initializing HTTP server on port 8082
Initializing broadcast thread
Spawning background conversion process
Starting recording
Starting websockets thread
Starting HTTP server thread
Starting broadcast thread
192.168.123.148 - - [04/May/2016 04:32:52] "GET / HTTP/1.1" 301 -
192.168.123.148 - - [04/May/2016 04:32:52] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:32:52] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:32:52] code 404, message File not found
192.168.123.148 - - [04/May/2016 04:32:52] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.148 - - [04/May/2016 04:33:04] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:33:04] "GET /jsmpg.js HTTP/1.1" 200 -
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3/dist-packages/ws4py/manager.py", line 310, in run
    if not ws.once():
  File "/usr/lib/python3/dist-packages/ws4py/websocket.py", line 305, in once
    if not self.process(b):
  File "/usr/lib/python3/dist-packages/ws4py/websocket.py", line 362, in process
    self.close(s.closing.code, s.closing.reason)
  File "/usr/lib/python3/dist-packages/ws4py/websocket.py", line 176, in close
    self._write(self.stream.close(code=code, reason=reason).single(mask=self.stream.always_mask))
  File "/usr/lib/python3/dist-packages/ws4py/websocket.py", line 243, in _write
    self.sock.sendall(b)
BrokenPipeError: [Errno 32] Broken pipe

192.168.123.148 - - [04/May/2016 04:34:34] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:34:34] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:34:34] code 404, message File not found
192.168.123.148 - - [04/May/2016 04:34:34] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.148 - - [04/May/2016 04:34:39] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:34:39] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:36:19] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:36:19] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:36:20] code 404, message File not found
192.168.123.148 - - [04/May/2016 04:36:20] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.148 - - [04/May/2016 04:40:43] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:40:43] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:41:00] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:41:00] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:41:01] code 404, message File not found
192.168.123.148 - - [04/May/2016 04:41:01] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.148 - - [04/May/2016 04:41:01] code 404, message File not found
192.168.123.148 - - [04/May/2016 04:41:01] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.148 - - [04/May/2016 04:41:19] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:41:19] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:43:12] code 404, message File not found
192.168.123.148 - - [04/May/2016 04:43:12] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.148 - - [04/May/2016 04:43:12] "GET /index.html HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:43:13] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.148 - - [04/May/2016 04:43:13] code 404, message File not found
192.168.123.148 - - [04/May/2016 04:43:13] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.142 - - [04/May/2016 04:46:56] "GET / HTTP/1.1" 301 -
192.168.123.142 - - [04/May/2016 04:46:56] "GET /index.html HTTP/1.1" 200 -
192.168.123.142 - - [04/May/2016 04:46:57] "GET /jsmpg.js HTTP/1.1" 200 -
192.168.123.142 - - [04/May/2016 04:47:00] code 404, message File not found
192.168.123.142 - - [04/May/2016 04:47:00] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.142 - - [04/May/2016 04:47:00] code 404, message File not found
192.168.123.142 - - [04/May/2016 04:47:00] "GET /favicon.ico HTTP/1.1" 404 -
192.168.123.142 - - [04/May/2016 04:47:01] code 404, message File not found
192.168.123.142 - - [04/May/2016 04:47:01] "GET /favicon.ico HTTP/1.1" 404 -

waveform80, 2018-03-01:
My best guess on this (as I couldn't replicate) was something to do with routing (grey page is the symptom seen when the video stream isn't available - commonly experienced by those using pistreaming over the Internet who've forwarded port 8082 but forgotten 8084). Anyway, I'm just having a spring-clean closing old (most likely abandoned) tickets - do feel free to re-open if this is still an issue for you!

closed 2018-03-01 by waveform80