waveform80/pistreaming #11: No longer works in iOS

markjerz, 2016-12-12:
Hi,

This no longer seems to work in iOS (I think since I upgraded to 10.) 

Any thoughts? I don't see any error messages client or server side so at a bit of a loss.

Thanks.

waveform80, 2017-02-03:
I'm afraid I've got no Apple products so this isn't something I can test. I'm getting a fair bit of e-mail regarding this, but all I can say at the moment is that this *might* be something to do with Apple doing something weird with websockets. Until I can get some concrete information though, I'm afraid I'll just have to mark pistreaming as "doesn't work with modern iOS devices" (oh well, bang goes the claim that it works on nearly everything ... still, goes to show just how bloody complex getting something to work on nearly all browsers actually is!)

WayneKeenan, 2017-05-08:
I was getting the following error in Safari on iOS and Mac

```WebSocket connection to 'ws://192.168.1.100:8084/' failed: Invalid HTTP version string: HTTP/1.0```

I have a fix incoming...

wujunze, 2017-11-27:
When do you combine this feature?  @waveform80

waveform80, 2018-03-01:
I'd forgotten about this, but there's some relevant discussion in #21: looks like I was waiting for someone to confirm a simpler fix would work on iOS. Given it's hung around though I'll just commit the simpler fix as it doesn't seem to affect anything on Linux/Windows/Android.

commit 0f676569 (2018-03-01): Fix #11, close #21

A simple fix for iOS (hopefully! I can't test it myself): use HTTP/1.1
as the version for the websocket response. Also changed avconv to ffmpeg
since Debian's now gone back to that (and it appears avconv isn't
present in Stretch).
closed 2018-03-01 by waveform80