waveform80/pistreaming #13: Broken Pipe Error

wardjk, 2017-01-14:
I frequently encounter "BrokenPipeError: [Errno 32] Broken pipe". Initially, I didn't know if the problem was with stdin or stdout but have sometimes seen an exception raised with the thread running the websocket server which indicates I have a problem with stdout. I considered changing the read bytes from 512 to 1024 but I don't know the implications of it. I read that doing so lead to problems in other cases. My understanding is that pipes are limited to 65K and image frames are significantly larger. Running at 320x240 @ 24fps can survive a long time but I can quickly get a broken pipe if running at 30fps. I'm using code adapted from server.py which is called from another class so I have other threads and timer events that complicate the issue and I likely have a timing problem. I’m also not using camera.start_recording() but instead am using camera.capture_continuous() so that I can modify images before sending but I don’t suspect this as contributing to the problem. I’m guessing that the stdout pipe and the broadcast rate aren’t playing well together. Any idea of what might be going on here?

waveform80, 2018-03-01:
Sorry I never got back to you on this - it's not a problem I've encountered before. However, I've recently had cause to do a bit of reading on synchronous and asynchronous network protocols, and how buffering interacts with them, and I think there's a subtle bug (or bad bit of design) in pistreaming's BroadcastThread which I think relates to this (and which has come up once before).

The issue is in the section you mention: the infinite loop that reads the ffmpeg output (`self.converter.stdout.read(512)`) and broadcasts it over the connected websocket(s). Originally that read a full 1024 bytes but that turned out to cause problems when the scene was largely static: the compression was sufficiently high that it took several frames to fill a full kilobyte and thus the clients stalled out waiting for the next frame. Lowering the read buffer size worked around that, but introduced another issue: sending tons of broadcast packets because now each was never more than 512 bytes in size. What's really needed is to tell the loop to read as much as is available *right now* but not to wait for a specific buffer size. In other words, the `read1` method.

I'm going to tweak the loop with this change and close this ticket as I'm just doing a spring clean and closing out old tickets that are probably abandoned. However, if you're still seeing the issue do feel free to re-open this!

closed 2018-03-01 by waveform80

wardjk, 2018-03-01:
Thanks. Your insight gave me ideas for other things to try.

commit 7a09dae9 (2018-03-01): Fix #13

Use a bigger read buffer, but use read1 to ensure early return if the
read buffer isn't yet full.