waveform80/pistreaming #14: No module name http.server?

creepandpeep, 2017-01-26:
Hi,

Followed instructions, but upon running server.py:

```
~/pistreaming $ ./server.py
Traceback (most recent call last):
  File "./server.py", line 12, in <module>
    from http.server import HTTPServer, BaseHTTPRequestHandler
ImportError: No module named http.server
```

I'm sure it's something obvious, but I'm a beginner. . . .could you help me out?

magicmad, 2017-01-26:
You need to use python 3, since python 2.7 does not have the HTTPServer.
starting it with "python3 server.py" should work.

creepandpeep, 2017-01-26:
wow - always the simple things :) works perfectly - stream is looking ace as well - thanks for the quick help !

closed 2017-01-26 by creepandpeep

SufyanParkar, 2020-06-15:
> You need to use python 3, since python 2.7 does not have the HTTPServer.
> starting it with "python3 server.py" should work.

Not working for me

sargentpanda, 2021-10-20:
yeah nothing is working for me

DarioDiGulio, 2022-06-22:
If your package named http, change this name. Working for me

Chdlopezes, 2023-02-05:
> If your package named http, change this name. Working for me

Yeah, I had this problem even with Python 3, so I change the name of my module, which was called "http". After the change, it works. Thank you!

zhc341272, 2023-02-05:
来信已收到，谢谢，吼吼！O(∩_∩)O~