waveform80/pistreaming #15: How would you go about making this accessible over the internet?

creepandpeep, 2017-01-29:
Stream is working perfectly on local network - if I wanted to make it accessible over the internet, I take it forwarding ports on a router isn't going to cut it for this?

waveform80, 2017-01-30:
It'll need a little more than that, but not much since it's just using TCP for everything. Firstly you'll need to forward two ports: 8082 for the HTTP connection, and 8084 for the WS connection. Secondly you'll need to tweak the code a bit so that the URL for the WS connection points to the router's *external* IP rather than the Pi's address...

If you look at `index.html` you'll see `ws://${ADDRESS}/` somewhere in there. That's the URL for the video stream. ADDRESS is substituted in `do_GET` (in `server.py`) for the socket's own address (see [`getsockname`](https://docs.python.org/3/library/socket.html#socket.socket.getsockname)). But that'll be internal to the LAN the Pi is on. You could simply hard-code that as your external address (or hostname - it doesn't have to be an IP), or you could do something a bit more nuanced to try and calculate your external address (e.g. Google "what's my IP?").

Anyway, I think that should be enough to get it working over the internet. As to how secure the code is ... no idea. I haven't put any thought into security on this; nothing's encrypted, there's no authentication, and the error checking is minimal.

creepandpeep, 2017-01-30:
Thank you for the reply, will give it a go later. As for security, or error
checking etc I'm not too worried for the circumstance I'm going to be using
it in, just want something quick and simple to begin with.

I'm just playing around really,  learning as I go...  got it streaming
 from embedded within a  WordPress post and looking slick :)  I know there
are pre made wp plugins for exactly this, but there's no fun in that and
they aren't  as flexible and styling is awkward.  I've also  put it in a
systemd service so it's simple to start and stop remotely and fits in with
another thing I'm doing (display o tron headless pi server control) .

Thanks again for your reply,

creepandpeep, 2017-01-30:
That's done it, replacing the address variable with hardcoded domain name and ports forwarded.  got a friend to test remotely and the quality/speed is apparently fantastic.

Thanks again!

closed 2017-01-30 by creepandpeep

waveform80, 2017-01-31:
Great stuff! Glad to see it's a simple as I thought. Don't know about the quality being great - this is only MPEG1 - but it's good enough for a demo.

The display-o-tron is something I need to have a play with (had one sitting in a drawer for a while). However, I did throw something together for pimoroni's pantilthat recently (if you've got one, checkout the pantilthat branch on this repo :). Anyway, have fun!

creepandpeep, 2017-01-31:
well, 640x480 picture was apparently very smooth and very fast - speaking with a friend on the phone and my lips were syncd to my voice.  I realise this is only one person connecting in, but over a home broadband connection which I later realised was also uploading a huge file via torrent at the same time, it was fantastic.

creepandpeep, 2017-01-31:
- and you should definitely play with the displayotron, it's great!  Accidentally ordered 2, ended up buying another pi to use them both at the same time :)