waveform80/pistreaming #18: Different resolution does not work

AuspeXeu, 2017-04-21:
Hello,
setting width to 1296 and height to 730 gives me an image like this.

[image]

aureliencousin, 2017-04-27:
Same problem here, I tried different resolutions and they all seem to have encoding problems. The only one working so far for me is the default one (640*480)

adrianorob, 2017-05-08:
I think is something related to byte size.

Like the standard starts with 640x480. So you its better to use 1280x760 if you want to increase it.

waveform80, 2017-09-10:
Sorry for the long delay replying; there's a big clue as to why this is happening in the console output:

```
Initializing HAT
Initializing camera
Initializing websockets server on port 8084
Initializing HTTP server on port 8082
Initializing broadcast thread
Spawning background conversion process
Starting recording
/usr/lib/python3/dist-packages/picamera/encoders.py:544: PiCameraResolutionRounded: frame size rounded up from 1296x730 to 1312x736
  width, height, fwidth, fheight)))
Starting websockets thread
Starting HTTP server thread
Starting broadcast thread
192.168.80.198 - - [10/Sep/2017 11:31:01] "GET /index.html HTTP/1.1" 200 -
192.168.80.198 - - [10/Sep/2017 11:31:02] "GET /styles.css HTTP/1.1" 200 -
192.168.80.198 - - [10/Sep/2017 11:31:02] "GET /jsmpg.js HTTP/1.1" 200 -
```

Unencoded output from the Pi's camera module is always in 32x16 blocks and thus resolutions selected must be a multiple of these blocks. The rounding warning in the output provides the resolution that was rounded to. If you use that resolution instead (in this case 1312x736) it'll work happily (well, more or less - MPEG1 requires quite a lot of bandwidth and processing power at that resolution).

waveform80, 2018-03-01:
Closing as I *think* this is satisfactorily answered - do feel free to re-open if it's still not working for you!

closed 2018-03-01 by waveform80