waveform80/pistreaming #22: Cannot open video stream on the web

adrianorob, 2017-05-08:
Hi all, 

Made the setup and everything is working fine in my local wireless connection. 

I have opened the port 8082 on port forwarding in my router, I can reach the server remotely and see it response. However, I dont get the video streaming, I dont know what should I do to get working on web. Its something related to router port forwarding or something like that? Could someone give any clue?

Its even strange because I can see the request on my Pi log, and it shows any error.

Thanks in advance.

WayneKeenan, 2017-05-09:
have a look at the startup messages...

```
Initializing websockets server on port 8084
Initializing HTTP server on port 8082
```

adrianorob, 2017-05-09:
Ok, should I enable the port 8084 as well? 

I did that and still is not showing the image, only the "loading...". 

Could you please be more specific?

WayneKeenan, 2017-05-09:
more specific, but you asked for "any clue" :)

Which os/platform (& version) are you using for the client ?

adrianorob, 2017-05-09:
Sorry is because I'm kinda of newbie in this world. :)

In my Pi is Raspbian:
```
~ $ cat /etc/os-release
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
```
On browsers, any browser is not working, in my local connection it works perfectly. However, when I try to watch the streaming from the web outside my local connection doesnt work, it shows the site and the loading without showing the video.

I enable the port forwarding in my router for ports 8082 and 8084, but it dont work. 

I think it might be something related to port forwarding with my router.

WayneKeenan, 2017-05-09:
Can you confirm you use exactly the same setup  (OS & browser combination) when on your Wifi and also when trying to access from outside your home network via the internet?

Your router rules should allow TCP (as opposed to UDP) traffic in both directions.

Open your browsers JavaScript console and paste here any errors you see.

adrianorob, 2017-05-09:
MacOS X El Capitan
Browser Chrome 

Both when on my Wifi and outside.

My router is configured allowing TCP for both directions.

And on my browser I dont see any console error. :/

WayneKeenan, 2017-05-09:
ok, it's not your router.

The server is serving up a local ip to your client connecting via the internet... I have a fix.

WayneKeenan, 2017-05-09:
Have a look at [this].(https://github.com/WayneKeenan/pistreaming/commit/9116848280d4279308085192f7144205bd69aabe)

Basically change the part of the line (51) that looks like this:

`self.request.getsockname()[0]
`

with this:

`"' + window.location.hostname +'"
`

adrianorob, 2017-05-09:
Hey @WayneKeenan , I will try right now, but makes total sense. Looking the network requests on chrome browser it shows the Pi local IP address not my network address.

adrianorob, 2017-05-09:
Its working!!!!! Thanks @WayneKeenan for the quick and awesome help!!!

Would you merge this fix into master?

WayneKeenan, 2017-05-09:
I'm not able todo that, ping @waveform80

waveform80, 2017-09-09:
Ahh, this actually came up in #15 but there I just suggested hard-coding the link to the websocket. You can probably tell I'm not a regular web-developer as I wasn't aware `window.location` was no longer a simple string! That's a very nice trick and I'll certainly add it in :)

commit 4e391f1b (2017-09-09): Fix #22

Many thanks to @WayneKeenan for a great window.location suggestion!
commit a21ffc0a (2017-09-25): Fix #22 on master

Fixed on pantilthat and then completely forgot to backport it!
closed 2017-09-09 by waveform80