waveform80/pistreaming #27: setting auto exposure

crosofg, 2017-09-02:
How to set the exposure automatically for low light?

waveform80, 2017-09-03:
Depends how dark the scene is. If there's low level artificial illumination, the default "auto" exposure is usually okay, but you can always choose "night" instead. You may also want to decrease the framerate to permit the AGC a bit more leeway in the exposure speed (see [Exposure Time](http://picamera.readthedocs.io/en/release-1.13/fov.html#exposure-time) in the picamera docs for more information).

As to how you should modify the script: simply set your required parameters after the camera is initialized (e.g. just after resolution and framerate are configured). It's also worth noting the exposure mode can be changed in the middle of streaming, but the framerate parameter cannot be (although you can cheat and use the framerate_delta parameter to modify the framerate on the fly, within certain limits).

waveform80, 2018-03-01:
Closing as answered - do feel free to re-open if you need more detail!

closed 2018-03-01 by waveform80