waveform80/pistreaming #28: Adafruit Neopixel Sticks not coloring correctly.

darkwaters2501, 2017-09-07:
I have 2 different types of the Adafruit Neopixel Sticks:
        Adafruit 2868 NeoPixel Stick RGBW.
        Adafruit 1426 NeoPixel Stick RGB.
Neither of which work exactly as expected. the color and pixel assignments are messed up.
I've found a partial fix if I add ether of these two lines at around line 21 in server.py file.

        For RGBW array I added:

hat.light_mode(hat.WS2812)
hat.light_type(hat.GRBW)

        Or for the RGB array I added:

hat.light_mode(hat.WS2812)
hat.light_type(hat.GRB)

This fixes most of the problems I've been getting, but for some reason the second and third pixels (1, 2) on the RGBW, or the second, third and forth pixels (1, 2, 3) on the RGB are still not being assigned their correct colors, and seem to be a little mixed up with each other.
Unfortunately I don't  know enough about code to find the reason for this and correct it.

waveform80, 2017-09-09:
Hmmm, confirmed. I could swear this worked last time I played with it (although that was quite a while ago) - I wonder if something's changed in the pantilthat library since then? I'm also getting the second and third RGB (but not W) pixels missing with an RGBW strip (which is the only one I've got), although they work when set individually.

Incidentally HAT initialization happens down in [main() at the bottom of the file](https://github.com/waveform80/pistreaming/blob/pantilthat/server.py#L190) and that already calls `light_mode`, although it doesn't currently call `light_type` (and that does indeed seem to, partially, fix things).

Anyway, I'll have a quick play and see if I can figure out what's causing the missing LEDs.

waveform80, 2017-09-09:
Okay, does seem like `light_type` definitely needs calling but the issue with the second and third LEDs not lighting is definitely an issue with the upstream pantilthat library (pistreaming is making the relevant calls, but nothing happens). The reason (I think - although I haven't tested this yet) is they're incorrectly attempting setting a white pixel in [set_pixel_rgbw](https://github.com/pimoroni/pantilt-hat/blob/master/library/pantilthat/pantilt.py#L290) (the `index` is incorrectly calculated, and it'd be redundant anyway as it's set down in [set_pixel](https://github.com/pimoroni/pantilt-hat/blob/master/library/pantilthat/pantilt.py#L336)). I'll file a PR for them in a bit.

closed 2017-09-09 by waveform80

waveform80, 2017-09-10:
Pixel issue reported upstream as pimoroni/pantilt-hat#13

commit b109290c (2017-09-09): Fix #28