waveform80/pistreaming #32: TLS Implementation

hornerj99, 2018-02-28:
It would be quite useful to be able to use HTTPS and WSS.  I'm pretty new to python and the LAMP stack in general, so I wasn't able to convert your server script myself.

Thank you!!

waveform80, 2018-03-01:
Hmm ... I'm keen to keep the script as simple as possible as it's not meant to be a "full product", but a base recipe for people to clone and hack on themselves. That said, security's always a good thing to have and if the implementation is sufficiently simple (doesn't complicate the understanding of the script and/or introduce tons of tricky options) then I'd be happy to merge such a PR. Unfortunately I haven't got time to hack on picamera or pistreaming right now (might have some next month depending on how things pan out, but picamera will be the priority then).

So, if you or anyone else fancies trying this and whipping up a PR: keep it as simple & basic as possible (don't bother with tons of options for cipher selection, key-exchange, etc. - just secure defaults with a couple of options to select the cert & key), and I'll certainly consider merging it!

hornerj99, 2018-03-02:
Great, thanks.  If/when I implement this, I'll get back in touch.  If someone else beats me to it, even better.  :)