waveform80/pistreaming #34: Error Running on Raspberry

541380000, 2018-03-09:
Dear author:
    I am a Chinese college students. My English is very basic so please reply to me in simple English, if any.
    I have been using Raspberry to make something. I want to play the video that is captured by cameras on Raspberry in my web browser. I use your pistreaming, but I get the following error:

 Initializing camera
Initializing websockets server on port 8084
Initializing HTTP server on port 8082
Initializing broadcast thread
Spawning background conversion process
Traceback (most recent call last):
  File "server.py", line 184, in <module>
    main()
  File "server.py", line 153, in main
    output = BroadcastOutput(camera)
  File "server.py", line 103, in __init__
    shell=False, close_fds=True)
  File "/usr/lib/python3.4/subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.4/subprocess.py", line 1457, in _execute_child
    raise child_exception_type(errno_num, err_msg)
 FileNotFoundError: [Errno 2] No such file or directory: 'ffmpeg'

I have been working on my project for a long time and it is the deadline ten days later. I have write all the other codes, but I just cannot get the video. Could you please help me?
If possible, I can send you a handmade gift in my hometown as a gift.
Thanks!

waveform80, 2018-03-09:
No need for gifts! You probably just need to install ffmpeg:

```
$ sudo apt install ffmpeg
```

I should probably change the README on the front page as `ffmpeg` used to be `avconv` and was in the `libav-tools` package (currently mentioned in the README), but that changed in Raspbian Stretch (as Debian upstream moved back to supporting ffmpeg rather than the avconv fork).

541380000, 2018-03-10:
Dear author:
I have solved this problem after installing ffmpeg through git clone.
Thanks for your help!

waveform80, 2018-03-10:
Glad to hear it!

closed 2018-03-10 by waveform80

reveriel, 2018-04-03:
It seems that `ffmpeg` has been replace by `avconv`.  

If you run `apt show libav-tools`, you will see:

```
This package replaces the 'ffmpeg' command-line tool that was provided 
in earlier distroreleases.
```

My suggestion is just replace the `ffmpeg` in file `server.py:92` with `avconv`. It works fine on my Pi3.

waveform80, 2018-04-11:
Back in Jessie that was true (avconv replaced ffmpeg) but the avconv fork of the ffmpeg project seems to have fallen out of favour upstream and ffmpeg is now back in Stretch as the "preferred" variant (the replacement message isn't present in more recent versions).