waveform80/pistreaming #36: Initializing server.py

disilva1993, 2018-06-27:
Hello, when i want to run the script this appears, do you know what can it be?
Thanks :)

pi@qwerty:~/pistreaming $ python3 server.py
Initializing camera
Initializing websockets server on port 8084
Initializing HTTP server on port 8082
Initializing broadcast thread
Spawning background conversion process
Traceback (most recent call last):
  File "server.py", line 184, in <module>
    main()
  File "server.py", line 153, in main
    output = BroadcastOutput(camera)
  File "server.py", line 103, in __init__
    shell=False, close_fds=True)
  File "/usr/lib/python3.4/subprocess.py", line 859, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.4/subprocess.py", line 1457, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'ffmpeg'

Billwilliams1952, 2018-06-28:
It seems you don’t have ffmpeg installed.

waveform80, 2018-10-04:
Yes, I should amend the installation instructions to include ffmpeg now that avconv seems to have fallen out of favour. Just a mo...

waveform80, 2018-10-04:
Fixed :)

closed 2018-10-04 by waveform80