waveform80/pistreaming #41: Audio Enhancement Suggestion !!!

yasersakkaf, 2019-01-04:
Hi,
Good work there. I needed to know how could I enhance your repo in order to stream the Video along with its audio???

lwilli, 2019-10-03:
I'm also very interested in supporting an audio source for pistreaming. 

I've tried changing the `ffmpeg` command in `server.py` to include replacing the audio stream with a separate source (https://superuser.com/a/277667/539627), but it doesn't work and just results in weird green stripes as the video or video with no audio. Any help would be greatly appreciated, as audio is an important part of my video streaming use case (and I would imagine it would be for others as well). Thanks!

Hamdi8, 2021-12-10:
Hello, 
Streaming video works veryy well, iv tried to add the audio , but ut doesn't work !
help please !!

zhc341272, 2021-12-10:
来信已收到，谢谢，吼吼！O(∩_∩)O~