waveform80/pistreaming #43: Taking stills

BrianP6, 2019-02-15:
Hello Dave,
Your PiStreaming is working superbly :-) thank you.

I would like to be able to take a still image/photo, would that be possible?
Using Raspistill doesn't want to work, presumably because other services have snaffled the resource.
I don't mind if the video stream flickers/pauses while the configuration modes are changed in order to get a higher resolution than 640x480 - in fact for a still image it is preferable.

Many thanks,
Brian.

BrianP6, 2019-02-23:
Just to add to this one. I've been reading in more and by modifying main() in the while loop for camera.wait_recording(1).

```python
while True:
                camera.wait_recording(1)
                if (os.path.isfile("/home/pi/camphoto")):
                    os.remove("/home/pi/camphoto")
                    my_file = open('/home/pi/camphoto.jpg', 'wb')

#                    camera.stop_recording()
#                    camera.resolution = (2592, 1944)
                    camera.capture(my_file, format="jpeg", use_video_port=True)
#                    camera.resolution = (WIDTH, HEIGHT)
                    my_file.close()

                    f = open("/home/pi/camphoto.done", "a")
                    f.close()
					
#                    camera.start_recording(output,'yuv')
```
The use of 'camphoto' files are notifications from one program to another, in this case if 'camphoto' is present, remove it then take a photo.

This works like a charm, but only works at 640x480 - as long as I don't change the resolution.

If I remove the comments to stop recording, increase the resolution, take the photo, put the resolution back and restart recording then it fails.
The photo does get taken and at the correct resolution, but pistreaming then fails.
From the output it appears to complain at BroadcastOutput.write stating that the file is closed.

I assume it's because there are other threads running while I've got the camera recording stopped that is causing the issue here, wondered if there's a way around it.

Many thanks,
Brian.

galed3e3, 2019-03-07:
Following. ..interested in doing similar.

rshom, 2021-09-21:
Also interested. I am trying to implement a button to the webpage that would take a picture.