waveform80/pistreaming #56: Autostart at reboot

DivineJimmi, 2021-02-05:
Hey guys,

is there an easy way to autostart the script at reboot? already tried the crontab.

`@reboot python3 /home/pi/pistreaming/server.py`
but this is not working.

rainripples, 2021-02-05:
Try 
@reboot /usr/bin/python3 /home/pi/pistreaming/server.py

crontab is very touchy about paths 

also make sure you “sudo crontab -e” to edit

DivineJimmi, 2021-02-05:
@rainripples This isnt working for me. Its still not starting at boot.

Notnagel, 2022-12-15:
There are always multiple ways to start scripts on the raspberry pi. Maybe check general sources on how to start python scripts on startup.

Personally I use rc.local
`sudo nano /etc/rc.local`
add line
`python3 /home/pistreaming/server.py` before `exit 0`
make it executable
`chmod +x /etc/rc.local`
reload systemd
`systemctl daemon-reload`
start your rc local
`systemctl start rc-local`

you can check the status of this service with
`systemctl status rc-local`

in many tutorials the systemd part is missing as it was, from my understanding, not neccessary in older debian versions